sparkfun/SparkFun-KiCad-Libraries #71: New v8 Search / Filter box limits part discovery

nseidle, 2024-03-19:
In v7, the searching the word 'SparkFun Push' allowed discovery of all push button switches because the search box included the file names ("SparkFun-Switch.pretty" was included). In v8, the file names are not searched, so only the FPs that have the word sparkfun in the keyword/description box is included in the search results.

I put 'SparkFun' in front of nearly all my searches to remove the kicad default libs, but this is hindering v8 use. This is a problem for symbol search as well.

Possible solutions:

* Add the keyword SparkFun to all FPs and symbols.
* Make users not import/use default KiCad libs/symbols, just search without the word SparkFun
* Allow import of default libraries, make user wade through 

@PaulZC I'm leaning towards adding the keyword but it feels brute force to me. Have you got any ideas?

[image]

[image]

And symbols as well:
[image]

PaulZC, 2024-03-19:
@nseidle : are you on 8.0.0 or 8.0.1? I _think_ it is behaving correctly for me in 8.0.1? But my libraries are still in v7 format...

nseidle, 2024-03-19:
Interesting. I'm on 8.0.1. I haven't converted or re-saved any lib (still in v7). I just did the standard adding of KiCad libraries to the end of the defaults:

[image]

How are you installing things?

PaulZC, 2024-03-20:
Here's what I see in 8.0.1:

In the schematic editor:

[image]

In the symbol editor:

[image]

Ah. Interesting. Searching for "sparkfun" doesn't find the PowerSymbol symbols. With no filter, it's there:

[image]

Add "sparkfun" as the filter and it's not there:

[image]

Everything goes weirdly non-alphabetic too:

[image]

PaulZC, 2024-03-20:
MicroMod is missing too. I think this only applies to MicroMod and PowerSymbols? Ah, MicroMod doesn't contain any symbols.

How am I installing things? With Preferences \ Manage Symbol Libraries. I use the folder button to add them all in one go:

[image]

Same for the footprint editor. Preferences \ Manage Footprint Libraries and the folder button:

[image]

Oh... Is it because PowerSymbols doesn't reference any SparkFun footprints? If it did, it would find them...

I'm with you on adding SparkFun to the Keywords. Sounds like a job for Reg Ex... Watch this space...

PaulZC, 2024-03-20:
OK. That's better:

[image]

PR is [here](https://github.com/sparkfun/SparkFun-KiCad-Libraries/pull/72)

nseidle, 2024-03-20:
I added keywords to the footprints.

closed 2024-03-20 by nseidle