sparkfun/SparkFun-KiCad-Libraries #110: Add courtyards around 0.1" PTH header footprints

sfe-SparkFro, 2024-08-20:
First off, all footprints should really have a courtyard to ensure DRC catches any accidentally overlapping parts.

Second, if a user wants to solder headers onto a board, there needs to be clearance for the headers, which are 0.1" wide. It's easy to forget this when designing a tight layout, such as the instance below (the outer 2x 0402 footprints would interfere with a header)

[image]

Just need to add courtyard rectangles 0.1" wide like so:

[image]

nseidle, 2024-08-28:
How about a much closer courtyard that would avoid collision with the standoff but still prevent components?

[image]

[image]

Also, I've added screws and nuts for 4-40 and M3.

[image]

[image]
[image]

SFE-Brudnerd, 2024-08-28:
My only argument against a tighter courtyard like what Nate suggests is if a user desires to populate with 0.1" headers then the black plastic could interfere.

nseidle, 2024-08-28:
Good point. We could add the outline to layer User1:

[image]

It will pass collision errors but make the designer aware of the possible issue.

Good enough?

SFE-Brudnerd, 2024-08-28:
WFM!

PaulZC, 2024-08-28:
My vote would be F.Fab. I've been using User.1 for milling. I submitted a PR recently to move all User1 text to F.Fab. I'd prefer not to start using it again for things like this...

PaulZC, 2024-08-28:
https://github.com/sparkfun/SparkFun-KiCad-Libraries/pull/106#issuecomment-2254102679

nseidle, 2024-08-28:
Talking this over with @sfe-SparkFro, we elected to keep the rectangular courtyard so that it does throw an error, and then the designer can exclude the error if they're ok with the collision.

[image]

Rectangular courtyards will land in a minute on main.

closed 2024-08-28 by nseidle