sparkfun/SparkFun-KiCad-Libraries #117: Part request - PRT-13054

owlyowl, 2024-09-04:
Hi I was wondering if it's possible to add this to the library please?

nseidle, 2024-09-04:
Sure! Merged now. You can find it under SparkFun-Connectors/Raspberry_Pi* But please check my work. I haven't used this part in a design yet so let me know if you see any issues. It's largely just a combination of built-in KiCad parts.

[image]

[image]

[image]

closed 2024-09-04 by nseidle

owlyowl, 2024-09-04:
> <sparkfun/SparkFun-KiCad-Libraries/issue/117/issue_event/14136107957@
> github.com>

Hi Nathan!
Thanks so much!!