sparkfun/SparkFun-KiCad-Libraries #120: Pin Header Footprints, Locked Versions Missing

thestumbler, 2024-09-24:
It looks like all the staggered-hole pin header footprints from the original SparkFun Eagle libraries, as written about in this [tutorial post](https://www.sparkfun.com/tutorials/114) by @lewispg228  way back in 2008, didn't survive the transition to KiCad libraries, with the lone exception of the 1x03 pin header. Any particular reason these were dropped?

I discovered you can open the old Eagle libraries in KiCad and a wide assortment of staggered-hole pin header footprints are available that way.

This is just an observation.  But assuming it wasn't an intentional decision to drop these footprints when you moved forward to KiCad libraries, I would say it is a suggestion as well.

nseidle, 2024-09-24:
Hello!

> Any particular reason these were dropped?

I should come up with a really good conspiracy theory, but the real answer is that we are moving over FPs only when we use them (it's work to clean them up and get them to our KiCad standards). And I just haven't designed anything with locking yet.

But, I went ahead and added 30 down to 3 of the locking FP (does 2 lock? 1 is silly), and 20 down to 3 of symbols. They should be merged into main. Let me know if I messed anything up. 

[image]
[image]

thestumbler, 2024-09-25:
Sounds good to me. BTW, as I was looking at these last night, I was struck by the fact that pin 1 of these footprints, indeed all pins, are not on the centerline of the connector pattern. That is by design, of course, the very definition of staggered. However, I might suggest that having pin 1 only not staggered could be a prudent thing to do without compromising the reason behind  staggering the pins in the first place. 

As is, the placement origin of the part is not the centerline of the pin 1 hole. One could be tempted to place the part on the board 5 mils offset from its intended location. I could certainly see myself making that blunder. In 99% of the cases that probably won't matter at all. 

So, if you whipped out all of these using a script, I might suggest considering such a modification (pin 1 doesn't get staggered). If you did all these by hand, I'd say leave it alone. 

Also I'll note that the original library had both 1x and 2x pin headers with staggering. Again, if you made these by script, I'd think making the corresponding 2x footprints also. Otherwise no big deal. It's trivial to take the 1x footprints and whip up a corresponding 2x as needed. 

Thanks for taking care of of this so quickly.

nseidle, 2024-09-25:
Me, [scripts](https://xkcd.com/1319/)? It was manual but not bad. 

I totally agree pin 1 should be on grid to aid locating.

2x pin headers will be left for a PR, or when somebody here needs them.

nseidle, 2024-09-25:
Pin 1 now on grid. Good idea! Again, let me know if anything looks odd.

commit 8cc46f90 (2024-09-24): Add locking PTH headers

Fix #120
closed 2024-09-25 by nseidle