sparkfun/SparkFun-KiCad-Libraries #127: Add voltage and current rating to regulators

sfe-SparkFro, 2024-10-10:
We commonly add the voltage and current rating of regulators in schematics as external text like so:

[image]

I think it would make sense to add those ratings to the regulators themselves, similar to the capacitors in this screenshot.

PaulZC, 2024-10-11:
It's a good suggestion... I included similar on the DC-DC converters:

[image]

[image]

[image]

nseidle, 2024-10-16:
I approve of all this. I haven't gotten a chance to get to it, so if you PR it, I'll make it so.

PR #142 (2024-11-13): Add parameters for voltage regulators
Fixes #127

Only addressed symbols in SparkFun-Regulator
closed 2024-11-13 by nseidle